santos-hacker-clube/aprenda_micropython #22: Controls > Arrays

ferauche, 2022-03-07:
Realizar o exemplo conforme https://github.com/arduino/arduino-examples/blob/main/examples/05.Control/Arrays/Arrays.ino deve ser criada uma pasta Controls no mesmo nível de Basics e Digital, e os arquivos desta tarefa devem estar dentro da pasta Controls\Arrays

gabrielzy18, 2022-03-08:
Prof, eu acho que o item ForLoopIteration acabou respondendo esse tbm

ferauche, 2022-03-09:
Acho que a forma como implementou o ForLoopInteration ficou ingual ao Arrays. Então renomei o diretório ForLoopIteration, para Arrays, e o ForLoppInteration deve ser implementado usando a função range, com os paramêtro indicando os pinos dos leds do pino 21 ao 23, depois outro de range de 25 a 27 já que o 24 não faz parte da sequência de leds.